Yet-Another-Software-Suite/YAGSL-Example #21: v2 Json Generator issues

BenFRC5147, 2023-02-20:
There are a few issues with the v2 Json Generator as it stands with the current version of the library

In all instances of 'absoluteEncoderOffset',  there is a typo that generates the json as 'absolueEncoderOffset' (missing the t in absolute)

In module properties, the json entry 'freespeedrpm' doesn't exist, can only be 'angleMotorFreeSpeedRPM' according to the build error when I try to build

In all modules, when inverting, it prepends the module name (frontleft, backright, etc) to the angle or motor line, reading backleftangle. The build error indicates the only acceptable fields to be angle and drive, so the frontleft, etc need to be removed. I didn't test to see if it worked with encoders being inverted, but it didn't say that they were an option when I was viewing the stacktrace.

And thats all I noticed with my program. Don't have a robot to test on now, but when I made these three changes to my json files, it built and ran a simulation!

Dylan-Watson, 2023-02-20:
That's what I get for writing this at 4 in the morning! A couple of other issues that I'll log here so I don't forget about them:
1) freespeedrpm should be hidden and auto-calculated based on the motor type selected (NEO: 5676, Falcon: 6800)
2) Add an "inches" label to the wheel diameter
3) Max speed units are ft/s

Thanks for taking a look at v2 and finding the issues! I'll try to fix it up tomorrow

BenFRC5147, 2023-02-20:
Also, I noticed that when trying to copy/paste, It constantly resets my highlights, my assumption is that this is occurring when it overwrites the data in the output field with the same data, if there is a way to check that no changes have occurred and stop it from updating/add a copy to clipboard button, that should resolve that as well

Dylan-Watson, 2023-02-20:
I did have one and I don't know where it went! I'll take a look at that as well.

BenFRC5147, 2023-02-21:
Looks like all the changes worked! Still no copy to clipboard button, but I am now able to highlight the file and it doesn't refresh and clear my highlights, which accomplishes the same purpose. Thank you!

Dylan-Watson, 2023-02-22:
Clipboard button added!