Yet-Another-Software-Suite/YAGSL-Example #99: Sparkmax Failover will never happen

TrialNTerror, 2023-11-29:
The getRelativeEncoderPosition method will always return the absolute encoder position as they are not separate in SparkmaxSwerve.java. Additionally absoluteEncoder will never equal null to swap over to the relative encoder. 

So it wont fail over. And the Dashboard says they are always identical regardless if both work or not.
[image]

TrialNTerror, 2023-11-29:
The absolute encoder value wont change if it comes unplugged. Would that be something that could be used?

TrialNTerror, 2023-11-29:
Ok Darn thank you

Technologyman00, 2023-11-30:
What if it was used for failover? I feel like that is the main reason.