Yet-Another-Software-Suite/YAGSL-Example #117: Navx Gyro Direction Won't Invert

weerobots, 2024-01-05:
First time user of YAGSL, working with 2024 preliminary example.
No robot but downloading to Roborio 1 with Navx mounted in MXP connector.
Tried _mxp as well as spi
Gyro always increases CW plus regardless of inversion true or false.
swervedrive.json is 
{
  "imu": {
    "type": "navx_spi",
    "id": 0,
    "canbus": null
  },
  "invertedIMU": true,
  "modules": [
    "frontleft.json",
    "frontright.json",
    "backleft.json",
    "backright.json"
  ]
}

Simulation works and runs the auto trajectory

Technologyman00, 2024-01-05:
Hello,

How are you reading the gyro always increases cw regardless of the inversion?

weerobots, 2024-01-05:
Hi,On the Shuffleboard Smart Dashboard tab. It shows as navx-sensor[4]John

weerobots, 2024-01-05:
My apologies - it does read plus in a ccw direction with a "false" for imu inversion. Not sure why I thought it didn't previously. Pretty sure I did try it both ways.
Only difference seems to be a power cycle.
Normally navx imus require inversion for yaw.

weerobots, 2024-01-05:
Closing - see previous comment

closed 2024-01-05 by weerobots