Yet-Another-Software-Suite/YAGSL-Example #157: [yaw/pitch/roll imu] imu orientation bug causes field centric bugs

Experance, 2024-01-29:
Different orientation of the imu causes field centric drive to not work properly. If you end up needing to review, our discussion was here: https://discord.com/channels/176186766946992128/1194445461852008498/1200599365434216508.

To replicate: Mount imu (Pigeon2 in my case) perpendicular to the robot, with it oriented right side up (so if you look at it, the words aren't upside down or on its side). When driving with the robot, field centrically works for just driving, but when driving while rotating, it does not act field centrically.

MarshallTappen, 2024-02-02:
I have a similar issue on our robot. I will try rolling back the changes from the ternary IMU change on Friday.

Ocean-Moist, 2024-02-16:
I am also experiencing this bug.

hardy-ethan, 2025-04-22:
Likely related to https://github.com/BroncBotz3481/YAGSL#333