Yet-Another-Software-Suite/YAGSL-Example #196: Cubic scaling makes max velocity unobtainable if not traveling in a cardinal direction

978-9-17-637879-3, 2024-04-01:
And it gets worse the closer you get to 45/135/225/315 degrees

Problem:
What happens is that the joystick inputs get cubed.

This allows you to reach maximum velocity when going in a cardinal direction, because the joystick reaches anywhere [0,1] before cubing, and thus [0,1] after cubing

However, if not going in a cardinal direction, your input is limited to [0, cos(arctan(xInput))] for x-axis and [0, sin(arctan[yInput])] for y-axis. 

Let's demonstrate the problem with an example: traveling 45 degrees. this is bounded to [0, ~0.707] from the above formula. if we cube .707, we get a shockingly low maximum power of 35.4%!!!

Solution:
Cube the magnitude of the joystick input, not the individual axes. The magnitude is [0,1] for any angle, so it does not face the same issue as raw axes.

978-9-17-637879-3, 2024-04-01:
I wrote some code that should fix this:

```
        double xInput = vX.getAsDouble();
        double yInput = vY.getAsDouble();

        Translation2d inputTranslation = new Translation2d(xInput, yInput);
        double magnitude = inputTranslation.getNorm();
        Rotation2d angle = inputTranslation.getAngle();

        double curvedMagnitude = Math.pow(magnitude, 3);

        double xVelocity = curvedMagnitude * angle.getCos() * swerve.maximumSpeed;
        double yVelocity = curvedMagnitude * angle.getSin() * swerve.maximumSpeed;
```

Technologyman00, 2024-04-01:
Awesome! I'm glad you were able to find this! Could you by chance show this in a simulation? Thank you!

978-9-17-637879-3, 2024-04-01:
I can try!

KangarooKoala, 2024-04-01:
For the ~~bean counters~~ instruction counters out there, below is an implementation that avoids any square root, division, or trig operations:
```java
// || (||v||² v) || = ||v||² ||v|| = ||v||³
// ||v||² is non-negative, so scaling by it preserves direction without flipping
double xInput = translationX.getAsDouble();
double yInput = translationY.getAsDouble();
double sqNorm = xInput * xInput + yInput * yInput;
xInput *= sqNorm;
yInput *= sqNorm;
// Do whatever with xInput and yInput
```
If you want to square magnitude instead of cubing you would need a square root, though.

978-9-17-637879-3, 2024-04-01:
> For the ~bean counters~ instruction counters out there, below is an implementation that avoids any square root, division, or trig operations:
>
> ```java
> // || (||v||² v) || = ||v||² ||v|| = ||v||³
> // [...]

Smart! I wish it was this easy to do with other curving methods (we're trying out lerp soon)

978-9-17-637879-3, 2024-04-01:
> Awesome! I'm glad you were able to find this! Could you by chance show this in a simulation? Thank you!

@Technologyman00 https://github.com/978-9-17-637879-3/YAGSL-Example/commit/adb5aa30bc2fb072b5a14bf4b1050841aee91d9c

978-9-17-637879-3, 2024-04-01:
I will say as well that my "fix" is not very efficient and it would likely better to just do the trig with pure math functions, as the math is quite simple. I'll be posting that version tomorrow (today??)

978-9-17-637879-3, 2024-04-01:
`      double xInput = translationX.getAsDouble();
      double yInput = translationY.getAsDouble();

      double magnitude = Math.hypot(xInput, yInput);
      
      // small magnitude impl. from Rotation2d
      double cos, sin;
      if (magnitude > 1e-6) {
        sin = yInput / magnitude;
        cos = xInput / magnitude;
      } else {
        sin = 0.0;
        cos = 1.0;
      }

      double curvedMagnitude = Math.pow(magnitude, 3);

      double xVelocity = curvedMagnitude * cos * swerveDrive.getMaximumVelocity();
      double yVelocity = curvedMagnitude * sin * swerveDrive.getMaximumVelocity();`

978-9-17-637879-3, 2024-04-02:
Students just tested this fix, works like a dream!

978-9-17-637879-3, 2024-04-11:
I feel this visual is a good indicator of the issue.
[image]
X and Y axes are joystick input, Z axis is commanded power.
Red is the faulty smoothing, green is the fixed smoothing.