Yet-Another-Software-Suite/YAGSL-Example #256: Update to 2025 Motor APIs

scarmain, 2024-12-02:
We have started the process of upgrading to 2025 with the beta, and realizing that there are significant changes in APIs for next season.

Known issues:
CANSparkMax is gone, and replaced with SparkMax.  This has a different way to configure the motor.

I tried an implementation of configuration, but I don't have a motor setup to try it yet.  See: https://github.com/FRC2832/RobotTemplate/blob/51d104d543c81a44597a7f8f8b871d3e7185c0bd/src/main/java/org/livoniawarriors/PidConstants.java#L204  (This whole class doesn't fully work, just trying to fix syntax errors.  Again, this is not tested, and Rev doesn't even have documentation on how to implement a PID yet, or examples in their examples project.  The best I can find is here: https://github.com/wpilibsuite/2025Beta/discussions/27)

scarmain, 2024-12-02:
Man, I need to fully read before posting...  I found an example: https://docs.revrobotics.com/brushless/revlib/revlib-overview/migrating-to-revlib-2025#before-2

thenetworkgrinch, 2024-12-02:
Already done, even did it before official docs came out on it 😂

https://github.com/BroncBotz3481/YAGSL-Example/tree/beta/src%2Fmain%2Fjava%2Fswervelib%2Fmotors

closed 2024-12-02 by thenetworkgrinch