Yet-Another-Software-Suite/YAGSL-Example #287: Issue getting absolute encoder to register properly

HolyArtist31656, 2025-01-07:
Code keeps crashing after enabling robot. Throws the "IF possible configure the encoder offset in the REV Hardware Client instead of using the absoluteEncoderOffset..." error message. Using MaxSwere modules with Neo drive motor, Neo 550 turn motor (both with SparkMax) and the Rev through bore encoder (REV-11-1271) plugged directly into the SparkMax for the turning motor. Link to the repository for the code I pushed (in case I'm just being dumb) is https://github.com/FRCteam2181/2025-Comp-Code

thenetworkgrinch, 2025-01-07:
That error message is no longer valid, use the JSON to set the offsets for REV Max Swerve.

HolyArtist31656, 2025-01-08:
I didn't realize that the configs had changed so much. I updated them but the code is still crashing saying that the SparkMax configs are receiving illegal parameters. I updated the code in the repository if looking at it helps. I also have the stacktrace if that would help.

HolyArtist31656, 2025-01-08:
Oh, sorry about that, it should be public now. Not sure why it wasn't already

HolyArtist31656, 2025-01-09:
The issue appears to be with the code for the integrated encoder in the swerve lib

thenetworkgrinch, 2025-01-09:
I am closing this discussion for now bc it's not determined to be a bug and is likely user-error. Please join the discord and ill be happy to debug there!

thenetworkgrinch, 2025-01-09:
https://discord.gg/5pYxSDnymr

closed 2025-01-09 by thenetworkgrinch