Yet-Another-Software-Suite/YAGSL-Example #299: Bug in TalonFX PIDF

prensing, 2025-01-15:
I believe this is a bug.

In TalonFXSwerve.java, line 205:
`  @Override
  public void configurePIDF(PIDFConfig config)
  {

    cfg.refresh(configuration.Slot0);
    cfg.apply(
        configuration.Slot0.withKP(config.p).withKI(config.i).withKD(config.d).withKS(config.f));
    //    configuration.slot0.integralZone = config.iz;
    //    configuration.slot0.closedLoopPeakOutput = config.output.max;
  }`

Should the last not be "withKF(config.f)" instead of "withKS()"?  This would explain some problems with tuning the PID of drive motors.

prensing, 2025-01-15:
OK, maybe this is not a problem, although I assume still a bug. I see that the actual FF is handled separately.

prensing, 2025-01-15:
Since I have a bug open, there is no way to fetch the drive motor feedforward object (so you can print it?).

prensing, 2025-01-15:
[Updated the issue title, since I am adding stuff.]

I have now figured out where the Drive FF is coming from. I feel this could be more transparent. I am pretty sure that a too aggressive FF is causing our (very light) chassis to drive poorly. However, I don't remember seeing anything about the FF in the documentation, particularly around tuning the PID. Tuning the drive PID really needs to start with tuning the FF.

thenetworkgrinch, 2025-01-15:
I am pretty sure the issue is resulting from the default maximums attainable speeds being set to 12, and it was 6. I set that to the maxSpeedMPS. I may change it to the maximum module speed if it fails the test. Please try the latest beta!

thenetworkgrinch, 2025-01-15:
https://github.com/BroncBotz3481/YAGSL-Example/commit/3a9dfac16cc5e7d9ed5777b58602fa16395b4a69

thenetworkgrinch, 2025-01-15:
If you would like to make additions to the documentation it is here, please feel free to PR in updates!
https://github.com/thenetworkgrinch/YAGSL-gitbook
I am closing this issue bc it seems resolved

closed 2025-01-15 by thenetworkgrinch