Yet-Another-Software-Suite/YAGSL-Example #321: `I` values cause heading to drift in simulation

CoffeeCoder1, 2025-02-05:
When an `I` value is set in the heading PID controller, the Robot pose heading will slowly drift, but the OdometryPose does not show this change. I haven't gotten a chance to test this on a real robot, so there's a chance that it's a bug in the heading control logic, but I feel like the gyro should correct for this, so it's probably something simulation-specific (maybe the integrator is putting out a small output which is turning the simulated robot but not the odometry?).

fletch3555, 2025-02-08:
You shouldn't need an `I` component on anything related to swerve, but I'm definitely curious about the source of the apparent drift

thenetworkgrinch, 2025-02-10:
This is exactly what you would see on a real robot i think.

closed 2025-02-10 by thenetworkgrinch

CoffeeCoder1, 2025-02-11:
I feel like this is still a problem that needs a bit of investigation, since, even if the heading drifts, the odometry should represent that change, right? There is a chance that I just don't understand how odometry works in YAGSL though. Also, in some further testing I have managed to reproduce this without any `I` component set (in simulation), it just happens at a much slower rate (just turn the robot a few times and wait each time to see if the Robot pose starts to drift at all - if you get it, it should just continue to turn very slowly).