Yet-Another-Software-Suite/YAGSL-Example #329: ADIS16470/ADIS16448 is flipped by default

rakosi2, 2025-02-21:
https://github.com/BroncBotz3481/YAGSL-Example/blob/b0473342f37cfbb64f46eb2381a61a7c42d01879/src/main/java/swervelib/imu/ADIS16470Swerve.java#L98

imu.getAngle() is already CCW+ , why is it `-1 * angle` here?

Same with ADIS16448, it is also already CCW+ and it is being inverted.

Could ADIS16470 get all three axes like ADIS16448 so that an offset can be set for different orientations?

Likely changes for the offseason since it will affect teams using these gyros

thenetworkgrinch, 2025-02-21:
It would be. AFAIK they read CCW- so I inverted them by default. I am not opposed to chnaging them in the offseason if you can cite where they are CCW+

Very few teams use these gyro's anymore so it makes sense that you caught it now

rakosi2, 2025-02-21:
[image]

rakosi2, 2025-05-07:
Looks like I read the graph for ADIS16448 wrong, it is upside down and we do not have one to test with.
Let me make a PR for ADIS16470 now that it is offseason. There is one axis that gets [Gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock) but added the other axes in case a team wants to orientate it in another axes.

closed 2025-05-08 by thenetworkgrinch