Yet-Another-Software-Suite/YAGSL-Example #333: YAGSL destroys Pigeon2 mount calibration

ajs256, 2025-03-09:
YAGSL factory-resets the IMU at code startup:
https://github.com/BroncBotz3481/YAGSL/blob/4016ee219003baabd6fe0ff481af86314ded9c42/swervelib/SwerveDrive.java#L281

For a Pigeon 2, this involves creating a new configuration object:
https://github.com/BroncBotz3481/YAGSL/blob/4016ee219003baabd6fe0ff481af86314ded9c42/swervelib/imu/Pigeon2Swerve.java#L92-L104

This configuration object includes the default values for *all* in-flash settings of the IMU, including the [mount calibration](https://v6.docs.ctr-electronics.com/en/stable/docs/tuner/pigeon-cal.html).

YAGSL **should not** destroy the calibration like this, or at least provide the user an opportunity to disable the reset. This would allow teams with IMUs mounted unconventionally to still take advantage of this important feature.

thenetworkgrinch, 2025-03-10:
I disagree unless you can show me where you can configure this in Tuner X. 

If this requires code in any way then teams can just to `(Pigeon2)SwerveDrive.getGyro().getImu()` and reconfigure it themselves.

ajs256, 2025-03-10:
@thenetworkgrinch 
> I disagree unless you can show me where you can configure this in Tuner X.

It's configured here: (screenshot from CTRE's docs, as I can't access a Windows machine right now)
[image]
Open the Pigeon2's page in Tuner X, then open the "Calibration" tab at the top of the screen.

> If this requires code in any way then teams can just to `(Pigeon2)SwerveDrive.getGyro().getImu()` and reconfigure it themselves.
Yes, but the issue is that the values are set by Tuner and most teams won't think they need to also put those values in code. 

I'll open a PR soon that replaces the call to reset the IMU altogether with one that just sends a `Pigeon2Features` object to the Pigeon2 (disabling the compass, but enabling the no-motion and temperature calibrations) to keep the current behavior wrt sensing but leaving the mount pose as-is.